m5stack/M5Atom #3: LEDDisplay example doesn't work

misosec, 2020-02-12:
Hello,
LEDDisplay example compiles without error, but led display stays blank.
Is there anything specific I have to do?

Windows 10
Arduino IDE version 1.8.10
I selected M5Stick-C board.

Thank you.

PaulMcGuinness, 2020-02-25:
I have same issue, but noticed the following compiler warning (incidentally the Easy Loader works fine):-

`C:\Users\Paul McGuinness\Documents\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.003

 #    pragma message "FastLED version 3.003.003"

                     ^

In file included from C:\Users\Paul McGuinness\Documents\Arduino\libraries\FastLED/FastLED.h:65:0,

                 from C:\Users\Paul McGuinness\Documents\Arduino\libraries\M5Atom\src/M5Atom.h:52,

                 from C:\Users\Paul McGuinness\Documents\Arduino\libraries\M5Atom\examples\Basics\LEDDisplay\LEDDisplay.ino:1:

C:\Users\Paul McGuinness\Documents\Arduino\libraries\FastLED/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"`

misosec, 2020-02-26:
@PaulMcGuinness, how can I use Easy Loader to upload LEDDidplay code? There is only Burn button without option to choose a file. Or am I missing something?
Thank You.

PaulMcGuinness, 2020-02-26:
Use this one:-

https://m5stack.oss-cn-shenzhen.aliyuncs.com/EasyLoader/M5Core/ATOM%20Matrix/EasyLoader_ATOM%20_Matrix.exe

misosec, 2020-02-26:
@PaulMcGuinness How can I choose what to upload?
Thanks again.

PaulMcGuinness, 2020-02-26:
You can't - It just loads a pre-compiled version of the example (which should be the same as the example here on Github that doesn't work!)

misosec, 2020-02-26:
@PaulMcGuinness I see.
Thanks anyway.

pm-pm, 2020-02-26:
@PaulMcGuinness
Thanks a lot for confirming my observations. 
I think they have other problems in china right now than a software update.
Let's hope the best for everyone.

ririfonfon, 2020-03-06:
change in LED_DisPlay.ccp line 103

 // xSemaphoreTake(_xSemaphore, portMAX_DELAY);
    xSemaphoreTake(_xSemaphore, 100);

Hades2001, 2020-06-02:
Thanks.We found this serious problem and have fixed it.

closed 2020-06-02 by Hades2001

anthonyn, 2021-03-16:
HI, I am seeing this in what I believe is the most up to date software.  What was the fix?