m5stack/M5Atom #5: LEDSet

pierreverbakel, 2020-02-23:
In the LEDSet.ino Example for the Atom as I used on the Atom Lite the following codeline should give RED but actualy produces Green:

 M5.dis.drawpix(0, 0xf00000); 

Using M5.dis.displaybuff() does work correctly. 
In the following lines it looks to me that something with the hardware had to be corrected by software:
void LED_Display::displaybuff(uint8_t *buffptr, int8_t offsetx, int8_t offsety)
...
            _ledbuff[x + y * 5].raw[1] = buffptr[2 + ((setdatax + x) % xsize + ((setdatay + y) % ysize) * xsize) * 3 + 0];
            _ledbuff[x + y * 5].raw[0] = buffptr[2 + ((setdatax + x) % xsize + ((setdatay + y) % ysize) * xsize) * 3 + 1];
            _ledbuff[x + y * 5].raw[2] = buffptr[2 + ((setdatax + x) % xsize + ((setdatay + y) % ysize) * xsize) * 3 + 2];
...

In M5.dis.drawpix() the Color is assigned directly leading to a swap of colors. 
Somebody who understands bitshifting could correct this in M5.dis.drawpix() maybe.
Thanks!

akita11, 2020-04-18:
I've encountered the same situation. It seems to be a bug...?

vany5921, 2020-04-26:
atom lite->SK6812
atom matrix->WS2812C

akita11, 2020-04-26:
Got it, it was my misunderstanding.
drawpix() is for ATOM lite
displayBuff() is for both ATOM matrix and lite
Right?

akita11, 2020-04-27:
Color settings in M5.dis.drawpix() seems to be wrong, that is different from M5.dis.displaybuff(DisBuff).
For example, M5.dis.drawpix(0, 0xff0000) flashes LED as green (that must be red), while M5.dis.displaybuff() with 0xff0000 in DisBuff[]  flashes red (correct).
In the source code, drawpix() is using the same function to displaybuff(), so I can't understand why color mismatch in drawpix() occurs. Here is the code I've tested.

`#include "M5Atom.h"
uint8_t DisBuff[2 + 5 * 5 * 3];

void setDisp(uint32_t color)
{
  DisBuff[0] = 0x05;
  DisBuff[1] = 0x05;
  for (int i = 0; i < 25; i++)
  {
    DisBuff[2 + i * 3 + 0] = color >> 16;
    DisBuff[2 + i * 3 + 1] = color >> 8;
    DisBuff[2 + i * 3 + 2] = color;
  }
  M5.dis.displaybuff(DisBuff);
}

void setup() {
  M5.begin(true, false, true);
  Serial.begin(115200);
}

void loop() {
  Serial.println("R1");
  setDisp(0xff0000); delay(500);
  setDisp(0x000000); delay(500);
  Serial.println("R2");
  M5.dis.drawpix(0, 0xff0000); delay(500);
  M5.dis.drawpix(0, 0x000000); delay(500);

  Serial.println("G1");
  setDisp(0x00ff00); delay(500);
  setDisp(0x000000); delay(500);
  Serial.println("G2");
  M5.dis.drawpix(0, 0x00ff00); delay(500);
  M5.dis.drawpix(0, 0x000000); delay(500);

  Serial.println("B1");
  setDisp(0x0000ff); delay(500);
  setDisp(0x000000); delay(500);
  Serial.println("B2");
  M5.dis.drawpix(0, 0x0000ff); delay(500);
  M5.dis.drawpix(0, 0x000000); delay(500);
}
`

akita11, 2020-04-28:
In my understanding, color settings for drawpix() should be changed to indicate GBR order, in order to match that in dispalybuf() to handle raw[1/0/2] order.

hrgraf, 2020-10-08:
Hi,

as it requires GBR order, the correct solution is to setup FastLED (line 5 in src/utility/LED_DisPlay.cpp) with:

    FastLED.addLeds<WS2812, DATA_PIN, GRB>(_ledbuff, LEDNumbre);

Then, no strange index rearrangement needed anymore in _displaybuff(), resp. _displaybuff()

For completenes, here is the complete fix patch:

diff --git a/src/utility/LED_DisPlay.cpp b/src/utility/LED_DisPlay.cpp
index 6b011c8..d24be8f 100644
--- a/src/utility/LED_DisPlay.cpp
+++ b/src/utility/LED_DisPlay.cpp
@@ -2,7 +2,7 @@

 LED_DisPlay::LED_DisPlay(uint8_t LEDNumbre)
 {
-    FastLED.addLeds<WS2812, DATA_PIN>(_ledbuff, LEDNumbre);
+    FastLED.addLeds<WS2812, DATA_PIN, GRB>(_ledbuff, LEDNumbre);
     _xSemaphore = xSemaphoreCreateMutex();
     _numberled = LEDNumbre;
 }
@@ -88,8 +88,8 @@ void LED_DisPlay::_displaybuff(uint8_t *buffptr, int8_t offsetx, int8_t offsety)
     {
         for (int y = 0; y < 5; y++)
         {
-            _ledbuff[x + y * 5].raw[1] = buffptr[2 + ((setdatax + x) % xsize + ((setdatay + y) % ysize) * xsize) * 3 + 0];
-            _ledbuff[x + y * 5].raw[0] = buffptr[2 + ((setdatax + x) % xsize + ((setdatay + y) % ysize) * xsize) * 3 + 1];
+            _ledbuff[x + y * 5].raw[0] = buffptr[2 + ((setdatax + x) % xsize + ((setdatay + y) % ysize) * xsize) * 3 + 0];
+            _ledbuff[x + y * 5].raw[1] = buffptr[2 + ((setdatax + x) % xsize + ((setdatay + y) % ysize) * xsize) * 3 + 1];
             _ledbuff[x + y * 5].raw[2] = buffptr[2 + ((setdatax + x) % xsize + ((setdatay + y) % ysize) * xsize) * 3 + 2];
         }
     }
@@ -129,8 +129,8 @@ void LED_DisPlay::displaybuff(uint8_t *buffptr, int8_t offsetx, int8_t offsety)
     {
         for (int y = 0; y < 5; y++)
         {
-            _ledbuff[x + y * 5].raw[1] = buffptr[2 + ((setdatax + x) % xsize + ((setdatay + y) % ysize) * xsize) * 3 + 0];
-            _ledbuff[x + y * 5].raw[0] = buffptr[2 + ((setdatax + x) % xsize + ((setdatay + y) % ysize) * xsize) * 3 + 1];
+            _ledbuff[x + y * 5].raw[0] = buffptr[2 + ((setdatax + x) % xsize + ((setdatay + y) % ysize) * xsize) * 3 + 0];
+            _ledbuff[x + y * 5].raw[1] = buffptr[2 + ((setdatax + x) % xsize + ((setdatay + y) % ysize) * xsize) * 3 + 1];
             _ledbuff[x + y * 5].raw[2] = buffptr[2 + ((setdatax + x) % xsize + ((setdatay + y) % ysize) * xsize) * 3 + 2];
         }
     }

yochy4671, 2020-11-08:
> Hi,
> 
> as it requires GBR order, the correct solution is to setup FastLED (line 5 in src/utility/LED_DisPlay.cpp) with:
> 
> ```
> FastLED.addLeds<WS2812, DATA_PIN, GRB>(_ledbuff, LEDNumbre);
> ```
> 
> Then, no strange index rearrangement needed anymore in _displaybuff(), resp. _displaybuff()
> 
> Please find the complete diff of the working solution attached.
> 
> [LED_DisPlay_diff.txt](https://github.com/m5stack/M5Atom/files/5351237/LED_DisPlay_diff.txt)

These solution are correct.

Please refer https://github.com/FastLED/FastLED/wiki/Rgb-calibration.
And try the sample sketch below.
```
#include "M5Atom.h"

void setup()
{
    M5.begin(true, false, true);
    delay(50);
    M5.dis.drawpix(0, CRGB::Red);
}

uint8_t FSM = 0;

void loop()
{
    if (M5.Btn.wasPressed())
    {

        switch (FSM)
        {
        case 0:
            M5.dis.drawpix(0, CRGB::Red);
            break;
        case 1:
            M5.dis.drawpix(0, CRGB::Green);
            break;
        case 2:
            M5.dis.drawpix(0, CRGB::Blue);
            break;
        case 3:
            M5.dis.drawpix(0, CRGB::White);
            break;
        default:
            break;
        }
        FSM++;
        if (FSM >= 4)
        {
            FSM = 0;
        }
    }

    delay(50);
    M5.update();
}
```

riraosan, 2021-05-08:
@yochy4671 
Hi.
>These solution are correct.
>
>Please refer https://github.com/FastLED/FastLED/wiki/Rgb-calibration.
>And try the sample sketch below.

`FastLED.addLeds<WS2812, DATA_PIN, GRB>(_ledbuff, LEDNumbre);`

I have confirmed that the above problem fixes the red and green swaps.
Thanks!👍

Tinyu-Zhao, 2021-09-02:
We have resolved this issue and will release it in the next version

closed 2021-09-02 by Tinyu-Zhao