m5stack/M5Atom #7: LED_Display.cpp  xSemaphoreTake & xSemaphoreGive correct

ririfonfon, 2020-03-12:
rewrite of LED_Display.cpp with xSemaphore Give & Take in the right order deletion of superfluous setBrightness (it's up to you to set the brightness value) 

`#include "LED_Display.h"

LED_Display::LED_Display(uint8_t LEDNumbre)
{
    FastLED.addLeds<WS2812, DATA_PIN>(_ledbuff, LEDNumbre);
    _xSemaphore = xSemaphoreCreateMutex();
    _numberled = LEDNumbre;
}

LED_Display::~LED_Display()
{
}

void LED_Display::run(void *data)
{
    data = nullptr;

    for (int num = 0; num < 26; num++)
    {
        _ledbuff[num] = 0x000000;
    }
    FastLED.show();

    while (1)
    {
        xSemaphoreTake(_xSemaphore, portMAX_DELAY);
        if (_mode == kAnmiation_run)
        {
            if ((_am_mode & kMoveRight) || (_am_mode & kMoveLeft))
            {
                if (_am_mode & kMoveRight)
                {
                    _count_x++;
                }
                else
                {
                    _count_x--;
                }
            }
            if ((_am_mode & kMoveTop) || (_am_mode & kMoveButtom))
            {
                if (_am_mode & kMoveTop)
                {
                    _count_y--;
                }
                else
                {
                    _count_y++;
                }
            }
            if ((_am_count != -1) && (_am_count != 0))
            {
                _am_count--;
                if (_am_count == 0)
                {
                    _mode = kAnmiation_stop;
                }
            }
             
            xSemaphoreGive(_xSemaphore);
            displaybuff(_am_buffptr, _count_x, _count_y);
            delay(_am_speed);
            delay(10);
            
        }
        else
        {
            xSemaphoreGive(_xSemaphore);
            delay(10);
        }
        FastLED.show();
        
    }
}

void LED_Display::animation(uint8_t *buffptr, uint8_t amspeed, uint8_t ammode, int64_t amcount)
{
    xSemaphoreTake(_xSemaphore, portMAX_DELAY);
    if (_mode == kAnmiation_run)
    {
        _mode = kAnmiation_stop;
    }
    _am_buffptr = buffptr;
    _am_speed = amspeed;
    _am_mode = ammode;
    _am_count = amcount;
    _count_x = _count_y = 0;
    _mode = kAnmiation_run;
    xSemaphoreGive(_xSemaphore);
}

void LED_Display::displaybuff(uint8_t *buffptr, int8_t offsetx, int8_t offsety)
{
    xSemaphoreTake(_xSemaphore, portMAX_DELAY);

    uint16_t xsize = 0, ysize = 0;
    xsize = buffptr[0];
    ysize = buffptr[1];

    offsetx = offsetx % xsize;
    offsety = offsety % ysize;

    int8_t setdatax = (offsetx < 0) ? (-offsetx) : (xsize - offsetx);
    int8_t setdatay = (offsety < 0) ? (-offsety) : (ysize - offsety);
    
    for (int x = 0; x < 5; x++)
    {
        for (int y = 0; y < 5; y++)
        {
            _ledbuff[x + y * 5].raw[1] = buffptr[2 + ((setdatax + x) % xsize + ((setdatay + y) % ysize) * xsize) * 3 + 0];
            _ledbuff[x + y * 5].raw[0] = buffptr[2 + ((setdatax + x) % xsize + ((setdatay + y) % ysize) * xsize) * 3 + 1];
            _ledbuff[x + y * 5].raw[2] = buffptr[2 + ((setdatax + x) % xsize + ((setdatay + y) % ysize) * xsize) * 3 + 2];
        }
    }
    xSemaphoreGive(_xSemaphore);
}

void LED_Display::setBrightness(uint8_t brightness)
{
    xSemaphoreTake(_xSemaphore, portMAX_DELAY);
    FastLED.setBrightness(brightness);
    xSemaphoreGive(_xSemaphore);
}

void LED_Display::drawpix(uint8_t xpos, uint8_t ypos, CRGB Color)
{
    if ((xpos >= 5) || (ypos >= 5))
    {
        return;
    }
    xSemaphoreTake(_xSemaphore, portMAX_DELAY);
    _ledbuff[xpos + ypos * 5] = Color;
    xSemaphoreGive(_xSemaphore);
}

void LED_Display::drawpix(uint8_t Number, CRGB Color)
{
    if (Number >= 25)
    {
        return;
    }
    xSemaphoreTake(_xSemaphore, portMAX_DELAY);
    _ledbuff[Number] = Color;
    xSemaphoreGive(_xSemaphore);
}

void LED_Display::clear()
{
    xSemaphoreTake(_xSemaphore, portMAX_DELAY);
    for (int8_t i = 0; i < 25; i++)
    {
        _ledbuff[i] = 0;
    }
    xSemaphoreGive(_xSemaphore);
}
`

PaulMcGuinness, 2020-03-30:
Regardless of using the quick semaphore fix or this one, I get loads of random coloured LED's on the bottom row whenever trying to scroll a message. There is no rhyme or reason to it... (no determinable pattern) - Has anyone else experienced this?

ririfonfon, 2020-03-30:
no ??

PaulMcGuinness, 2020-03-31:
Here's the code I wrote to scroll an IP address, let me know if it works for any of you?

`#include "M5Atom.h"

const unsigned char cMatrix[13][60] = {
  {
    0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00,
    0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 
    0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, // 1
    0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00,
    0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00
  },{                                                            
    0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, // 2
    0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0x00,0x00,0x00,
    0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff
  },{                                                            
    0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, // 3
    0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff
  },{                                                            
    0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00,
    0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff, // 4
    0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff
  },{                                                            
    0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0x00,0x00,0x00,
    0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00, // 5
    0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00
  },{
    0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0x00,0x00,0x00,
    0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00, // 6
    0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00
  },{                                                            
    0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, // 7
    0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00,
    0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00
  },{                                                            
    0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, // 8
    0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff
  },{                                                            
    0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff, // 9
    0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff
  },{                                                            
    0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00,
    0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0xff,0xff,0xff, // 0
    0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0xff,0xff,0xff,
    0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00
  },{                                               
    0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00,
    0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00,
    0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, // .
    0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00,
[... 49 lines cut ...]
      
         //   Serial.println("Char:" + String(myMsg.charAt(i)) + " loffset=" + String(lOffset) + " line=" + String(line) + " i2=" + String(i2));       

            for(int p=0;p<charW*colors;p++)
            {
              int p2=(i*charW*colors)+p; // Position in target array
              int px=cMatrix[i2][p+(line*colors*charW)];
              
           //   Serial.println("We will set aData[" + String(p2+lOffset+2) + "] to cMatrix[" + String(i2) + "][" + String(p+(line*colors*charW)) + "] = " + String(px));
              aData[p2+lOffset+2]=px;
            }
          }
      }
      
     
    
    
    M5.dis.animation((uint8_t *)aData, 500, LED_Display::kMoveLeft, screenW-5);
}

void loop()
{
    delay(500);
    M5.update();
}`

ririfonfon, 2020-03-31:
Hello,

here are the corrections, I'm working on vscode 

so I'm joining you *.cpp  you can change it to * .ino if necessary.

I had change the size of the table in figures to understand your code 

but the problem was not right in the size of the message

 

 
 
 
  
     
  
 
          
  
 
 
 

L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

#include "M5Atom.h"
const unsigned char cMatrix[13][77] = {

    {/* Line   000 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0x00,0x00,0x00, // 
     /* Line   001 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00, 0x00,0x00,0x00, // 
     /* Line   002 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0x00,0x00,0x00, // 1
     /* Line   003 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0x00,0x00,0x00, // 
     /* Line   004 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0x00,0x00,0x00},// 

    {/* Line   000 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00, // 
     /* Line   001 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, // 
     /* Line   002 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00, // 2
     /* Line   003 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, // 
     /* Line   004 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00},// 

    {/* Line   000 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00, // 
     /* Line   001 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, // 
     /* Line   002 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00, // 3
     /* Line   003 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, // 
     /* Line   004 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00},// 

    {/* Line   000 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, // 
     /* Line   001 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, // 
     /* Line   002 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00, // 4
     /* Line   003 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, // 
     /* Line   004 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00},//

    {/* Line   000 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00, // 
     /* Line   001 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, // 
     /* Line   002 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00, // 5
     /* Line   003 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, // 
     /* Line   004 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00},//

    {/* Line   000 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00, // 
     /* Line   001 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, // 
     /* Line   002 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00, // 6
     /* Line   003 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, // 
     /* Line   004 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00},// 

    {/* Line   000 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00, // 
     /* Line   001 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, // 
     /* Line   002 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, // 7
[... 25 lines cut ...]
     /* Line   004 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0x00,0x00,0x00},//

    {/* Line   000 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, // 
     /* Line   001 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, // 
     /* Line   002 */ 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00, // -
     /* Line   003 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, // 
     /* Line   004 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00},//

    {/* Line   000 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, // 
     /* Line   001 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, // 
     /* Line   002 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, // space
     /* Line   003 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, // 
     /* Line   004 */ 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00} // 

};

const int charW = 5;
    String myMsg = "111.222.333.444 ";

    int arraySize = msgLen * ((colors * charW * charH)+msgLen) ;

                aData[p2 + lOffset + 5] = px;
            }
        }
    }

PaulMcGuinness, 2020-03-31:
Hi - Same (if not worse) results. Your code tries to make the chars 5 pixels wide (I was deliberately trying to make it 4 pixels so that it took less of the display and made scrolling better).

I suspect I have a faulty unit as displaying different numbers produces different weird effects.

PaulMcGuinness, 2020-03-31:
Works! :-)

ririfonfon, 2020-03-31:
yes i saw you have to calibrate the length of the message

what i send to you works with xxx.xxx.xxx.xxx

but not with xxx.xx.x.xx

so I assume that:

int arraySize = msgLen * ((colors * charW * charH) + msgLen);

is not the right defenition

 

and yes i agree with you on the width of 4 pixel

I modified it to understand your function

 

https://www.avast.com/antivirus

ririfonfon, 2020-03-31:
Ok you find the correct fonction

Please you can return it.

 

https://www.avast.com/antivirus

PaulMcGuinness, 2020-03-31:
Ok.... You pointed me in the right direction, and this now works with 4 x 5 chars and any string:-

```
#include "M5Atom.h"
const unsigned char cMatrix[13][60] = {
  {
  0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00,
  0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, // 1
  0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00
  },{
  0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, // 2
  0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0x00,0x00,0x00,
  0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff
  },{
  0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, // 3
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff
  },{
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00,
  0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff, // 4
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff
  },{
  0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0x00,0x00,0x00,
  0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00, // 5
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00
  },{
  0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0x00,0x00,0x00,
  0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00, // 6
  0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00
  },{
  0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, // 7
  0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00
  },{
  0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, // 8
  0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff
  },{
  0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0xff,0xff,0xff, // 9
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff
  },{
  0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00,
  0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0xff,0xff,0xff, // 0
  0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00, 0xff,0xff,0xff,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00
  },{
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, // .
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0xff,0xff,0xff, 0x00,0x00,0x00
  },{
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00,
  0x00,0x00,0x00, 0xff,0xff,0xff, 0xff,0xff,0xff, 0x00,0x00,0x00, // -
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00
  },{
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, // [space]
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00,
  0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00, 0x00,0x00,0x00
  }
};

const String cChars = "1234567890.- ";
const int charW = 4; // Width of the character matrix char (4 leds)
const int charH = 5; // Height of the character matric char (5 leds)
const int colors = 3; // Number of tuples for each char (colors)
bool IMU6886Flag = false;

void setup()
{
  M5.begin(true, false, true);
  delay(50);
  String myMsg = "123 - 4 - 9.6.0";
  int msgLen = myMsg.length();
  int screenW = msgLen * charW;

  int arraySize = (msgLen * ((screenW * charH))) + 4 ; // Add 4 bytes for header
  unsigned char aData[arraySize]; // 3 colors, 4 pixels, 5 rows
  
  aData[0] = screenW;
  aData[1] = charH;

  for (int line = 0; line < charH; line++)
  {
    int lOffset = (line * msgLen) * colors * charW; // 3 colors, 4 pixels wide
    
    for (int i = 0; i < myMsg.length(); i++)
    {
      int i2 = cChars.indexOf(myMsg.charAt(i)); // Offset in cMatrix[i2]
      i2 = (i2 == -1) ? 12 : i2; // Space for unknown chars
      for (int p = 0; p < charW * colors; p++)
      {
      int p2 = (i * charW * colors) + p; // Position in target array
      int px = cMatrix[i2][p + (line * colors * charW )];
      aData[p2 + lOffset + 5] = px;
     }
    }
  }

  M5.dis.animation((uint8_t *)aData, 150, LED_Display::kMoveLeft, screenW - charW);
}

void loop()
{
  delay(150);
  M5.update();
}
```

Hades2001, 2020-06-02:
Thanks for the suggestion,We found this serious problem and have fixed it.

closed 2020-06-02 by Hades2001