m5stack/M5Atom #13: Deep sleep current too high

erniberni, 2020-05-01:
I have tested 4 Atom Lite with the deep-sleep sketch from the ESP32 library. I powered the modules directly with 3.3V and measured the power consumption in the 3.3V supply line. Nothing else was connected. 
2 of these modules show a deep-sleep current of 40mA, which seems much too high. The other two modules have a deep-sleep current of 4mA, which is quite acceptable.
Do you have an explanation for this?

JPFrancoia, 2020-08-27:
Hey, sorry to hijack your question. How did you power the board with 3.3V directly? I mean, to which pin did you input a 3.3V tension?

erniberni, 2020-09-05:
There is one connector of the female connector strip labeled with 3V3.

miag676, 2021-06-07:
Hi @erniberni, I'm thinking of using m5Atom Lite for my battery-powered project. I'm worried about the power consumption being too high, so I was wondering if you managed to find out how high is the power consumption in deep-sleep mode and why did you get different deep-sleep currents (40 mA and 4 mA)?

Best regards,
Mia

erniberni, 2021-06-07:
Hi @miag676,
unfortunately, I could never find out the real cause. I suspect that a component, e.g. the voltage regulator, was damaged when I used the Lite with a very poor voltage source. But this is only a guess. The quiescent current of 4mA seems to be the correct result. This value would be realistic if the quiescent current of the USB-serial converter is not optimised. If you really need a very low quiescent current then use a ESP32 WROOM module without anything else connected and a very good LDO.

linxcow, 2022-08-31:
I also have this M5 atom lite , i bought it for low power battery applications. ( product page : wearables?)
My deep sleep current is 9mA (rgb off) powered from 5v pin.
These are not recommended for long term battery applications unfortunately due to bad design ( high current LDO & RGB)

I als found this out after i bought this because there is no deepsleep current mentioned on the product page.
stay away for battery operated projects unless only for a few days. other esp32 boards can go as low as 40uA deepsleep

erniberni, 2022-08-31:
Thank you, I agree in total