m5stack/M5Atom #15: LED flicker icm with WiFi

pilipili-vr, 2020-05-28:
When using the LED display in combination with the WiFi radio (either client or AP mode), some LEDs flicker in random colors.

HawkCollective2890, 2020-06-30:
I have found the same thing with Bluetooth.

haakonnessjoen, 2020-07-13:
In M5Atom.cpp, inside `if ( DisplayEnable ) {`, add this line: `dis.setCore(1);`, before `dis.start();`

This puts the LED update code into core 1. Whereas wifi is handeled in core 0.

Seems to solve the issue, but I have not tested it extensively.

pilipili-vr, 2020-07-13:
Doesn't work for me. I also tried using Fastled separately without using the M5 library and running everything on core 1 (except for Wifi) but that also didn't help. It's a shame, because like this it is really unusable.

alexdmejias, 2020-07-30:
Happy to see that I'm not the only one having this issue. Sad to see that this is an issue

felmue, 2020-08-07:
Hi @haakonnessjoen 
thank you for sharing this information. Moving the LED update code into core 1 works for my M5AtomMatrix. I no longer see any random LED flicker.
Thanks
Felix

DAKarp, 2020-10-25:
This fixes the flicker for me:

```
  M5.begin(/* SerialEnable: */ true, /* I2CEnable: */ false, /* DisplayEnable: */ false);

  M5.dis.setTaskName("LEDs");
  M5.dis.setTaskPriority(2);
  M5.dis.setCore(1);
  M5.dis.start();
```

pilipili-vr, 2020-10-29:
Also didn't work. I also tried different task priorities but didn't help, some (priority 4, 1 and 0) even gave completely nonfunctional display (all LEDs stay full red, but without flickering though)

closed 2022-07-19 by Tinyu-Zhao