m5stack/M5Atom #23: Can't upload to M5Atom

dpharris, 2020-07-30:
Hi --

I am having trouble uploading to either the M5Atom or the M5StickC.

I am running:
 MacBook Pro
 OSX 10.10.5
 Arduino IDE 1.8.13

I have tried: different baud rates, new cable, powered hub.  

M5Burner doesn't work, either.  

I am able to upload to a Uno, and a Tiva via Energia (Arduino IDE clone) via usb.  

Help! 

David

johnwargo, 2020-08-25:
So you're not providing any details about what happens when you try to do it, so I doubt anyone can help you. "I'm having trouble" and "doesn't work either" aren't enough information to act on. 

May I suggest that you post your questions, with detailed error messages, to the [Community Forum](https://community.m5stack.com/)?

Tinyu-Zhao, 2022-06-28:
Since your description is too vague and there are no compilation/error hints, I guess it is a driver problem, you can try to install a driver suitable for your computer system and try again

closed 2022-06-28 by Tinyu-Zhao