m5stack/M5Atom #28: LEDDisplay doesn't display anything

johnwargo, 2020-08-26:
The sample [LEDDisplay](https://github.com/m5stack/M5Atom/tree/master/examples/Basics/LEDDisplay) app has a typo:

```c
//    M5.dis.animation((uint8_t *)AtomImageData, 200, LED_DisPlay::kMoveLeft, 18);
    M5.dis.animation((uint8_t *)AtomImageData, 200, LED_Display::kMoveLeft, 18);
```

but even with that fix, the sketch runs on the device, but nothing appears on the display. 

I see that there's PRs attempting to fix related issues, any chance someone from M5Stack can look at all those PRs and fix the sample apps?

KodeMunkie, 2020-10-18:
+1 for this, should be an easy fix - looks like somebody renamed a method based on an M5Atom.h change but didn't test the code.

closed 2021-08-30 by Tinyu-Zhao

johnwargo, 2021-08-30:
@Tinyu-Zhao I see you closed it - why? Was this resolved? Have you merged the related pull requests? 

Standard GitHub protocol for closing issues says that you should explain why you're closing it and, if appropriate, link the issue to the action that fixed the issue.

Tinyu-Zhao, 2021-08-30:
> @Tinyu-Zhao I see you closed it - why? Was this resolved? Have you merged the related pull requests?
> 
> Standard GitHub protocol for closing issues says that you should explain why you're closing it and, if appropriate, link the issue to the action that fixed the issue.

https://github.com/m5stack/M5Atom/pull/20
https://github.com/m5stack/M5Atom/tree/master/examples/Basics/LEDDisplay
You can try it,If you have any other questions, please contact me.