m5stack/M5Atom #37: Atom_POE Ethernet2 vs Ethernet

supaplextor, 2021-05-08:
Hello, I have a handful of Atom_POE devices, and my first proof of concept was to use this code:

https://github.com/m5stack/M5Atom/tree/master/examples/ATOM_BASE/ATOM_PoE

The sample sketch relies on Ethernet2. It will not compile, returns an error "cannot declare variable server to be of abstract type EthernetServer.

I've attempted to poke around and use Ethernet.h; I'm unable to see that work as well. Can someone work on this and update from legacy code (Ethernet2 is end-of-life aka technical debt).

Thank you!

felmue, 2021-05-13:
Hello @supaplextor 

please have a look at a possible solution [here](https://community.m5stack.com/topic/3068/lan-module-w5500-with-poe-compilation-error/2) - this should allow the example to compile with Ethernet.h:

Thanks
Felix

supaplextor, 2021-05-16:
I might of goofed, but here's what I have at present.

supaplex@supaplex-N56JR:~/usr/src/github-by-user/m5stack/M5Atom/examples/ATOM_BASE/ATOM_PoE$ git diff -u
diff --git a/examples/ATOM_BASE/ATOM_PoE/ATOM_PoE.ino b/examples/ATOM_BASE/ATOM_PoE/ATOM_PoE.ino
index 4ccc449..2eefc8c 100644
--- a/examples/ATOM_BASE/ATOM_PoE/ATOM_PoE.ino
+++ b/examples/ATOM_BASE/ATOM_PoE/ATOM_PoE.ino
@@ -8,7 +8,7 @@
 */
 
 #include <SPI.h>
-#include <Ethernet2.h>
+#include <Ethernet.h>
 #include <FastLED.h>
 #include "index.h"
 
@@ -99,4 +99,3 @@ void loop() {
     Serial.println("client disconnected");
   }
 }
-

supaplex@supaplex-N56JR:~$ find . -name Ethernet.h 2>/dev/null 
./.arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/Ethernet/src/Ethernet.h
./usr/src/junk/EthernetWebServer/LibraryPatches/Ethernet/src/Ethernet.h
./Arduino/hardware/esp8266com/esp8266/libraries/Ethernet/src/Ethernet.h

supaplextor, 2021-05-17:
Thank you,

I Modified these sources accordingly:

- /home/supaplex/Arduino/ATOM_PoE/ATOM_PoE.ino
- /usr/local/bin/arduino-1.8.15/libraries/Ethernet/src/Ethernet.h
- /usr/local/bin/arduino-1.8.15/libraries/Ethernet/src/EthernetServer.cpp

closed 2021-05-17 by supaplextor