m5stack/M5Atom #39: ATOM CAN-example  don't work (also the EASYLOADER examples)

rudydevolder, 2021-05-17:
I tried everything; **even the EASYLOADER examples** to find out that these **don't work either!!!**, I own 6 CAN-bus boards and different ATOM5 and MATRIX modules. I tested them all. Always the same problem: when resetting the sender only 2 messages are send than hang... And the receiver keeps rebooting after the button is 1 time pressed, until the sender is reset, but  I don't get any further than that.
For the moment I'm still guessing the direction of the problem:
I tried to avoid conflicts by:
  1. Deleting all references to the display library: 
  2. Also disabling I2c: M5.begin(true,false,false); 
But everything to no avail.
I found out this M5-Atom CAN example is actually based on the same libraries and examples from: [](http://www.iotsharing.com/2017/09/how-to-use-arduino-esp32-can-interface.html)
So I guess it should work on other boards. 
Someone having any idea to search in what direction ?

slechtvalk200, 2021-09-24:
I have exactly the same problem, tried with arduino 1.8.10 and 1.8.15. tried on usb 3.0 and 2.0 port, all the same problem. any solution?

this line causes the problem : "ESP32Can.CANWriteFrame(&tx_frame);"

pgrueber, 2023-04-16:
I also have exactly the same problem, I've tried four M5atom CAN units, all have the same problem, any ideas how I can solve this?

commenting out this line alows the program to continue, not very helpful : "ESP32Can.CANWriteFrame(&tx_frame);"

pgrueber, 2023-04-16:
I found a solution that works for me, place this line “ESP32Can.CANStop();" after this line "ESP32Can.CANWriteFrame(&tx_frame);" and place this line "ESP32Can.CANInit();" before "ESP32Can.CANWriteFrame(&tx_frame);" . It would seem that sending more than three frames sequentially will cause the M5Atom CAN to lockup. Reinitializing the ESP32Can for each TX frame avoids this issue.

pgrueber, 2023-04-25:
Update, I'm using an M5Atom CAN to transmit a single message containing simulated Toyota speed information, the software runs fine for about 21minutes then the CAN chip stops transmitting, the software keeps running but no data on the CANBus. I’ve tried a few modules, but they all do the same thing.