m5stack/M5Atom #59: Compile error with the latest stable release 2.0.2 of arduino-esp32

akihitonoda, 2022-01-04:
The following simple sketch gets a compile error:

`#include <M5Atom.h>`
`void setup() {`
`}`
`void loop() {`
`}`

Error message includes:

> error: call of overloaded 'begin(int, int, int)' is ambiguous   Wire.begin(25, 21, 100000);
note: candidate: 'bool TwoWire::begin(int, int, uint32_t)'
     bool begin(int sda=-1, int scl=-1, uint32_t frequency=0); // returns true, if successful init of i2c bus
note: candidate: 'bool TwoWire::begin(uint8_t, int, int, uint32_t)'
     bool begin(uint8_t slaveAddr, int sda=-1, int scl=-1, uint32_t frequency=0);

Enviromnent:
Arduino IDE 1.8.19
arduino-esp32 2.0.2
M5Atom 0.0.7

Comments:
The ambiguity stated in the error message seems to be caused by the slave mode operation newly implemented in the Wire library in arduino-esp32.
The ambiguity can be avoided by adding ‘UL’ suffix to the third argument, the I2C clock frequency, of calling Wire.begin() as follows:
`Wire.begin(25,21,10000UL);`
The call of Wire.begin() appears in both M5Atom.cpp (23rd line) and MPU6886.cpp (35th line).

P.S. I’m a newbie to the OSS community, so I’m not sure whether I should report such an issue on the M5Atom repository or on the arduino-esp32 repository. Please forgive my awkward post.

https://github.com/m5stack/M5Atom/blob/a7792813484c831a5e8d5c0c78ad35a24cb9eed2/src/M5Atom.cpp#L23

fuyunfei, 2022-01-06:
Hi,  @akihitonoda   same problem here , did you fix it ?

akihitonoda, 2022-01-06:
Hi, @fuyunfei, thank you for your feedback.
Actually, I have downgraded the arduino-esp32 library to 2.0.0 because I don't need the latest version for now. 
It works fine for me.
If you need the latest version of the arduino-esp32 library, another option is editing two library files, M5Atom.cpp and MPU6886.cpp.
They call Wire.begin() function with three integer arguments, which cause the "ambiguous" error. 
This is due to the arduino-esp32 2.0.2, particularly, the newly added Wire.begin() function overload with four arguments, which is for "slave" operation on the I2C bus.
To avoid ambiguity, the third argument, the clock frequency of I2C, should be explicitly presented as the uint32_t type.
Simply adding a suffix 'UL' to the third argument resolves this problem.
https://github.com/m5stack/M5Atom/blob/a7792813484c831a5e8d5c0c78ad35a24cb9eed2/src/M5Atom.cpp#L23
This should be
`Wire.begin(25,21,100000UL);`
and 
https://github.com/m5stack/M5Atom/blob/a7792813484c831a5e8d5c0c78ad35a24cb9eed2/src/utility/MPU6886.cpp#L35
this should be
`Wire1.begin(25,21,100000UL);`

I hope this will help you.

fuyunfei, 2022-01-07:
wow, it's working  perfectly,  big help,  you saved my day! @akihitonoda

Tinyu-Zhao, 2022-03-12:
Thank you for your feedback, we have fixed our library and will be compatible with the latest version of the arduino-esp32 library in the next release.

closed 2022-03-12 by Tinyu-Zhao