m5stack/M5Atom #78: Failed to install library: 'M5Atom:0.1.1'. No valid dependencies solution found: dependency 'ATOM-ECHO' is not available

Herve3, 2023-09-20:
### Describe the bug

I am trying to install this library in Ardino IDE (2.2.2) but I received this error message:

Failed to install library: 'M5Atom:0.1.1'. No valid dependencies solution found: dependency 'ATOM-ECHO' is not available

Can not find anywhere this dependency.

Best regards

### To reproduce

1 . I am on ubuntu 20.04 LTS
2. I start a blank project with Ardino IDE 2.2.2
3. I go in manage library and try to install M5Atom library (see screenshot)

### Expected behavior

I expect the library dependency be on an existing library.

### Screenshots

[image]

### Environment

- OS: Ubuntu 20.04
- IDE &IDE Version: Ardino ide 2.2.2

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Atom/issues?q=)
- [X] My report contains all necessary details

DavidZisky, 2023-09-27:
I had the same problem and I installed 0.1.0 library instead. It works fine

Gitshaoxiang, 2023-10-19:
fixed in the latest version

closed 2023-10-19 by Gitshaoxiang