m5stack/M5Atom #81: ATOM_CAN build fails for M5AtomS3

sonic74, 2024-06-04:
This repo is linked on https://docs.m5stack.com/en/atom/Atomic%20CAN%20Base which also has pin definitions for M5AtomS3 which implies that the ATOM_CAN example should work for M5AtomS3, but already building fails:

```
In file included from C:\Users\sven\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.1.1/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_reg.h:21,
                 from C:\Users\sven\AppData\Local\Temp\.arduinoIDE-unsaved202454-42800-1lh1i6q.yvqn\ATOM_CAN\CAN.c:38:
C:\Users\sven\AppData\Local\Temp\.arduinoIDE-unsaved202454-42800-1lh1i6q.yvqn\ATOM_CAN\CAN.c: In function 'CAN_init':
C:\Users\sven\AppData\Local\Temp\.arduinoIDE-unsaved202454-42800-1lh1i6q.yvqn\ATOM_CAN\CAN.c:169:29: error: 'DPORT_PERIP_CLK_EN_REG' undeclared (first use in this function); did you mean 'SYSTEM_PERIP_CLK_EN1_REG'?
     DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
                             ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\sven\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.1.1/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:75:67: note: in definition of macro '_DPORT_WRITE_PERI_REG'
 #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
                                                                   ^~~~
C:\Users\sven\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.1.1/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:88:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_SET_PERI_REG_MASK(reg, mask)   DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)|(mask)))
                                              ^~~~~~~~~~~~~~~~~~~~
C:\Users\sven\AppData\Local\Temp\.arduinoIDE-unsaved202454-42800-1lh1i6q.yvqn\ATOM_CAN\CAN.c:169:5: note: in expansion of macro 'DPORT_SET_PERI_REG_MASK'
     DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
     ^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\sven\AppData\Local\Temp\.arduinoIDE-unsaved202454-42800-1lh1i6q.yvqn\ATOM_CAN\CAN.c:169:29: note: each undeclared identifier is reported only once for each function it appears in
     DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
                             ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\sven\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.1.1/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:75:67: note: in definition of macro '_DPORT_WRITE_PERI_REG'
 #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
                                                                   ^~~~
C:\Users\sven\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.1.1/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:88:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_SET_PERI_REG_MASK(reg, mask)   DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)|(mask)))
                                              ^~~~~~~~~~~~~~~~~~~~
C:\Users\sven\AppData\Local\Temp\.arduinoIDE-unsaved202454-42800-1lh1i6q.yvqn\ATOM_CAN\CAN.c:169:5: note: in expansion of macro 'DPORT_SET_PERI_REG_MASK'
     DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
     ^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\sven\AppData\Local\Temp\.arduinoIDE-unsaved202454-42800-1lh1i6q.yvqn\ATOM_CAN\CAN.c:169:53: error: 'DPORT_CAN_CLK_EN' undeclared (first use in this function); did you mean 'DPORT_REG_CLR_BIT'?
     DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
                                                     ^~~~~~~~~~~~~~~~
C:\Users\sven\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.1.1/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:75:88: note: in definition of macro '_DPORT_WRITE_PERI_REG'
 #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
                                                                                        ^~~
C:\Users\sven\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.1.1/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:88:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_SET_PERI_REG_MASK(reg, mask)   DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)|(mask)))
                                              ^~~~~~~~~~~~~~~~~~~~
C:\Users\sven\AppData\Local\Temp\.arduinoIDE-unsaved202454-42800-1lh1i6q.yvqn\ATOM_CAN\CAN.c:169:5: note: in expansion of macro 'DPORT_SET_PERI_REG_MASK'
     DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
     ^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\sven\AppData\Local\Temp\.arduinoIDE-unsaved202454-42800-1lh1i6q.yvqn\ATOM_CAN\CAN.c:170:31: error: 'DPORT_PERIP_RST_EN_REG' undeclared (first use in this function); did you mean 'SYSTEM_PERIP_RST_EN0_REG'?
     DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_CAN_RST);
                               ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\sven\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.1.1/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:75:67: note: in definition of macro '_DPORT_WRITE_PERI_REG'
 #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
                                                                   ^~~~
C:\Users\sven\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.1.1/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:85:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_CLEAR_PERI_REG_MASK(reg, mask) DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)&(~(mask))))
                                              ^~~~~~~~~~~~~~~~~~~~
C:\Users\sven\AppData\Local\Temp\.arduinoIDE-unsaved202454-42800-1lh1i6q.yvqn\ATOM_CAN\CAN.c:170:5: note: in expansion of macro 'DPORT_CLEAR_PERI_REG_MASK'
     DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_CAN_RST);
     ^~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\sven\AppData\Local\Temp\.arduinoIDE-unsaved202454-42800-1lh1i6q.yvqn\ATOM_CAN\CAN.c:170:55: error: 'DPORT_CAN_RST' undeclared (first use in this function); did you mean 'DPORT_DATE_REG'?
     DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_CAN_RST);
                                                       ^~~~~~~~~~~~~
C:\Users\sven\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.1.1/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:75:88: note: in definition of macro '_DPORT_WRITE_PERI_REG'
 #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
                                                                                        ^~~
C:\Users\sven\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.1.1/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:85:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_CLEAR_PERI_REG_MASK(reg, mask) DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)&(~(mask))))
                                              ^~~~~~~~~~~~~~~~~~~~
C:\Users\sven\AppData\Local\Temp\.arduinoIDE-unsaved202454-42800-1lh1i6q.yvqn\ATOM_CAN\CAN.c:170:5: note: in expansion of macro 'DPORT_CLEAR_PERI_REG_MASK'
     DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_CAN_RST);
     ^~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\sven\AppData\Local\Temp\.arduinoIDE-unsaved202454-42800-1lh1i6q.yvqn\ATOM_CAN\CAN.c:175:40: error: 'CAN_TX_IDX' undeclared (first use in this function); did you mean 'TWAI_TX_IDX'?
     gpio_matrix_out(CAN_cfg.tx_pin_id, CAN_TX_IDX, 0, 0);
                                        ^~~~~~~~~~
                                        TWAI_TX_IDX
C:\Users\sven\AppData\Local\Temp\.arduinoIDE-unsaved202454-42800-1lh1i6q.yvqn\ATOM_CAN\CAN.c:180:39: error: 'CAN_RX_IDX' undeclared (first use in this function); did you mean 'TWAI_RX_IDX'?
     gpio_matrix_in(CAN_cfg.rx_pin_id, CAN_RX_IDX, 0);
                                       ^~~~~~~~~~
                                       TWAI_RX_IDX
C:\Users\sven\AppData\Local\Temp\.arduinoIDE-unsaved202454-42800-1lh1i6q.yvqn\ATOM_CAN\CAN.c:252:20: error: 'ETS_CAN_INTR_SOURCE' undeclared (first use in this function)
     esp_intr_alloc(ETS_CAN_INTR_SOURCE, 0, CAN_isr, NULL, NULL);
                    ^~~~~~~~~~~~~~~~~~~
C:\Users\sven\AppData\Local\Temp\.arduinoIDE-unsaved202454-42800-1lh1i6q.yvqn\ATOM_CAN\ATOM_CAN.ino: In function 'void setup()':
[... cut ...] ' was not declared in this scope
     CAN_cfg.tx_pin_id = GPIO_NUM_22;
                         ^~~~~~~~~~~
C:\Users\sven\AppData\Local\Temp\.arduinoIDE-unsaved202454-42800-1lh1i6q.yvqn\ATOM_CAN\ATOM_CAN.ino:50:25: note: suggested alternative: 'GPIO_NUM_42'
     CAN_cfg.tx_pin_id = GPIO_NUM_22;
                         ^~~~~~~~~~~
                         GPIO_NUM_42

Compilation error: 'DPORT_PERIP_CLK_EN_REG' undeclared (first use in this function); did you mean 'SYSTEM_PERIP_CLK_EN1_REG'?
```

Tinyu-Zhao, 2024-06-05:
Can you provide your program?

sonic74, 2024-06-05:
This is not my program, but yours: In the Arduino IDE just go to the menu File/Examples/M5Atom/ATOM_BASE/ATOM_CAN and try to build it for M5AtomS3

sonic74, 2024-06-05:
Might be a good idea to link to
https://github.com/m5stack/M5AtomS3/tree/main/examples/Unit/MiniCAN_TJA1051T
on the website. And also correct the table
https://static-cdn.m5stack.com/resource/docs/products/atom/Atomic%20CAN%20Base/pinMap-34359801-b635-49e7-8b00-61bf09e7b4d1.png
"ATOM LITE" is two times in there and AtomS3Lite is missing.

Tinyu-Zhao, 2024-06-17:
You can try this example

```cpp
#include <M5AtomS3.h>
#include "driver/twai.h"

// Pins used to connect to CAN bus transceiver:
#define RX_PIN 6
#define TX_PIN 5

// Interval:
#define TRANSMIT_RATE_MS 1000

#define POLLING_RATE_MS 1000

#define ARDUINO_USB_CDC_ON_BOOT 1

static bool driver_installed = false;

unsigned long previousMillis = 0;  // will store last time a message was send

void setup() {
    auto cfg = M5.config();
    AtomS3.begin(cfg);

    // Initialize configuration structures using macro initializers
    twai_general_config_t g_config = TWAI_GENERAL_CONFIG_DEFAULT(
        (gpio_num_t)TX_PIN, (gpio_num_t)RX_PIN, TWAI_MODE_NORMAL);
    twai_timing_config_t t_config =
        TWAI_TIMING_CONFIG_500KBITS();  // Look in the api-reference for other
                                        // speed sets.
    twai_filter_config_t f_config = TWAI_FILTER_CONFIG_ACCEPT_ALL();

    // Install TWAI driver
    if (twai_driver_install(&g_config, &t_config, &f_config) == ESP_OK) {
        Serial.println("Driver installed");
    } else {
        Serial.println("Failed to install driver");
        return;
    }

    // Start TWAI driver
    if (twai_start() == ESP_OK) {
        Serial.println("Driver started");
    } else {
        Serial.println("Failed to start driver");
        return;
    }

    // Reconfigure alerts to detect TX alerts and Bus-Off errors
    uint32_t alerts_to_enable = TWAI_ALERT_TX_IDLE | TWAI_ALERT_TX_SUCCESS |
                                TWAI_ALERT_TX_FAILED | TWAI_ALERT_ERR_PASS |
                                TWAI_ALERT_BUS_ERROR;
    if (twai_reconfigure_alerts(alerts_to_enable, NULL) == ESP_OK) {
        Serial.println("CAN Alerts reconfigured");
    } else {
        Serial.println("Failed to reconfigure alerts");
        return;
    }

    // TWAI driver is now successfully installed and started
    driver_installed = true;
}

static void send_message() {
    // Send message

    // Configure message to transmit
    twai_message_t message;
    message.identifier       = 0x0F6;
    message.data_length_code = 4;
    for (int i = 0; i < 4; i++) {
        message.data[i] = 0;
    }

    // Queue message for transmission
    if (twai_transmit(&message, pdMS_TO_TICKS(1000)) == ESP_OK) {
        printf("Message queued for transmission\n");
    } else {
        printf("Failed to queue message for transmission\n");
    }
}

void loop() {
    if (!driver_installed) {
        delay(1000);
        return;
    }

    uint32_t alerts_triggered;
    twai_read_alerts(&alerts_triggered, pdMS_TO_TICKS(POLLING_RATE_MS));
    twai_status_info_t twaistatus;
    twai_get_status_info(&twaistatus);

    if (alerts_triggered & TWAI_ALERT_ERR_PASS) {
        Serial.println("Alert: TWAI controller has become error passive.");
    }
    if (alerts_triggered & TWAI_ALERT_BUS_ERROR) {
        Serial.println(
            "Alert: A (Bit, Stuff, CRC, Form, ACK) error has occurred on the "
            "bus.");
        Serial.printf("Bus error count: %lu\n", twaistatus.bus_error_count);
    }
    if (alerts_triggered & TWAI_ALERT_TX_FAILED) {
        Serial.println("Alert: The Transmission failed.");
        Serial.printf("TX buffered: %lu\t", twaistatus.msgs_to_tx);
        Serial.printf("TX error: %lu\t", twaistatus.tx_error_counter);
        Serial.printf("TX failed: %lu\n", twaistatus.tx_failed_count);
    }
    if (alerts_triggered & TWAI_ALERT_TX_SUCCESS) {
        Serial.println("Alert: The Transmission was successful.");
        Serial.printf("TX buffered: %lu\t", twaistatus.msgs_to_tx);
    }

    unsigned long currentMillis = millis();
    if (currentMillis - previousMillis >= TRANSMIT_RATE_MS) {
        previousMillis = currentMillis;
        send_message();
    }
}

```