m5stack/M5Atom #83: M5Atom/src/M5Atom.h:83:8: error: extended character “ is not valid in an identifier

mgrouch, 2024-06-11:
### Describe the bug

/home/runner/Arduino/libraries/M5Atom/src/M5Atom.h:83:8: error: extended character “ is not valid in an identifier
   83 | #error “This library only supports boards with ESP32 processor.”
      |        ^
/home/runner/Arduino/libraries/M5Atom/src/M5Atom.h:83:64: error: extended character ” is not valid in an identifier
   83 | #error “This library only supports boards with ESP32 processor.”
      |                                     

### To reproduce

try compiling

### Expected behavior

code need to compile

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Additional context

n/a

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Atom/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2024-06-12:
You should choose the wrong board.
https://docs.m5stack.com/zh_CN/arduino/arduino_board

Tinyu-Zhao, 2024-06-12:
Fixed.

closed 2024-06-12 by Tinyu-Zhao