m5stack/M5Atom #85: Atom Echo example PlayMusic.ino error in Atom.h

PaulskPt, 2024-10-06:
### Describe the bug

Using Arduino IDE v2.3.3. When compiling the example PlayMusic.ino for the Atom Echo the compiler reported an error:

```
C:\Users\<Username>\Documents\Arduino\libraries\M5Atom\src/M5Atom.h:82:64: error: extended character ” is not valid in an identifier
   82 | #error “This library only supports boards with ESP32 processor.”.
```

 I consulted Microsoft CoPilot. This AI assistant immediately pointed to the error: CoPilot answered: ```The issue you’re encountering is due to the use of “smart quotes” (curly quotes) instead of straight quotes in your code. The compiler does not recognize these characters as valid in C++ code.```. I followed CoPilot's advise. Then the sample sketch compiled and ran without error.

### To reproduce

Just try to compile the example from [PlayMusic](https://github.com/m5stack/M5Atom/tree/master/examples/Echo/PlayMusic)

### Expected behavior

I expected an Arduino compile session without error!

### Screenshots

[image]

### Environment

- OS MS Windows 11 Pro
- IDE &IDE Version: Arduino IDE v2.3.3
- Repository Version: M5Stack M5Atom 0.1.2

### Additional context

May I also suggest some textual alteration in the example ```PlayMusic.ino```:
There are TMHO, some textual (not correct English) errors:
like "Spaker" instead of "Speaker" and "Spakeflag" instead of "Speakerflag".
For example:
line 82:    Serial.println("Init Spaker");
line 94:     if (Spakeflag) {

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Atom/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2024-10-08:
Sorry for the inconvenience. This is caused by not setting the latest release as the last release. I have fixed the problem and am waiting for the Arduino Library to be synchronized. If you are in a hurry, you can pull the latest code on GitHub to fix this problem.           
[image]

closed 2024-10-08 by Tinyu-Zhao

PaulskPt, 2024-10-08:
Thank you. I already corrected the small error. I just wanted you guys to know and later not to have my change be overwritten by an update that would continue to have the error. Thank you for your good work!